ArduPilot/ardupilot_wiki #621: Discourse server is incredibly slow

rmackay9, 2016-11-30:
It takes me 6 minutes to load the discourse servers: http://discuss.ardupilot.org/

For me this makes it nearly usable.

gmorph, 2016-12-02:
I have the same issue.  I think its the network routing.  See the picture below showing massive packet loss.
[image]

OXINARF, 2016-12-03:
@gmorph Are you always routed through there with that packet loss?

rmackay9, 2016-12-24:
For me, the biggest improvement came from clearing up the IP address conflict of my home's two routers (one wifi, one non-wifi) which I had connected in parallel.
Performance for me is now back to about 15seconds to load discourse.  It can take a couple of minutes now and then but mostly it's far better.
There's a plan to move to AWS (or similar) servers.

closed 2017-04-25 by rmackay9