ArduPilot/ardupilot_wiki #627: Plane SITL/MAVProxy Tutorial ：some codes need to be changed

nocternal, 2016-12-02:
[Plane SITL/MAVProxy Tutorial](http://ardupilot.org/dev/docs/plane-sitlmavproxy-tutorial.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

for new APM-Plane ,it's not
` param load ..\Tools\autotest\ArduPlane.parm`
it should be:
` param load ../Tools/autotest/default_params/plane-jsbsim.parm`

anyone cone to upodate this wiki page?

peterbarker, 2016-12-04:
@nocternal In the top-RHS of that page there's an "edit on github" link.

If you follow it - you can create a pull request to fix the issue, which would be awesome.

Are you happy to do that?

nocternal, 2016-12-05:
Yes. I create a PR which add 4 additons and 1 deletion, but it's weird that shows my modifcation(juhan-patch-1) has 138 addtions and 134 deletions... a lot of lines are deleted and add back automaticly.

khancyr, 2017-02-16:
can be closed

closed 2017-02-16 by nocternal