ArduPilot/ardupilot_wiki #637: Gimbal options not available

kilwater, 2016-12-10:
Hello, 
I'm trying to setup a simple 2D analog servo gimbal using APM 2.8/Mission Planner 3.1.41 with arduplane 3.7.1. 

When I'm connecting everything together (a11, a10 for signal, v+ and gnd from output line powered by BEC - jp1 removed)  - servo starts to buzz loudly and its not reacting to apm gyro movement.
I've tried 4 different servos. Same happens when I switch gimbal chanell to rc6,7,8 etc. 

Other thing - Mission Planner in Gimbal section, allows me only to select a chanel from a drop-down list , Servo Limits and reverse option.
 I cannot select Stabilise (greyed), Input Ch (greyed), Type (greyed), Angle Limits etc...

Can you help me out?

Peter
[image]

magicrub, 2016-12-10:
With old 8bit APM 2. x hardware for limited to old firmware. I see that
you've connected as arduplane v3.4 and this sounds like a great help
question for the forums discuss.ardupilot.org instead of the developer
feature request and to do list.

auturgy, 2017-10-15:
As above, this is a support request better suited for the forum.  Closing.

closed 2017-10-15 by auturgy