ArduPilot/ardupilot_wiki #638: LIDAR-Lite Rangefinder

pengguiyong2016, 2016-12-12:
[LIDAR-Lite Rangefinder](http://ardupilot.org/copter/docs/common-rangefinder-lidarlite.html) -how to use the i2c interface in the pixhawk? somebody can tell~~~~~

billbonney, 2016-12-12:
This is questions for the support forums. This issue list is for documentation issues.

see  http://discuss.ardupilot.org

closed 2016-12-12 by billbonney

rmackay9, 2016-12-12:
Bill's right about where support questions belong but an answer is below in any case.

@pengguiyong2016, by the way, there are multiple versions of the lidar-lite.  If you're trying to use v3 you'll find that it doesn't work with Copter-3.4.  The fix is in master though so you'll be able to use it with AC3.5 which will start beta testing in early January.

http://ardupilot.org/copter/docs/common-rangefinder-lidarlite.html

ak1801, 2017-05-14:
Hi Randy,

Is the beta testing complete for AC3.5? I am able to find AC3.5.0-rc5 under Firmware site, path: /Copter/beta/PX4.
Please confirm if I can use this firmware version to connect Garmin Lidar Lite v3 with Pixhawk using PWM.

Thanks.

rmackay9, 2019-01-16:
@meec1002317, I guess this is really more of a support question so it should probably go in [the forums](https://discuss.ardupilot.org/c/arducopter/copter-36).  The lidar lite v3 should work for [terrain following in manual modes](http://ardupilot.org/copter/docs/terrain-following-manual-modes.html) and [autonomous modes](http://ardupilot.org/copter/docs/terrain-following.html).