ArduPilot/ardupilot_wiki #640: Data Logs Documentation out of date

Peterrnold, 2016-12-22:
Hi
I am new to analysing data logs in Mission Planner and found that the web documentation seems out dated compared to the latest version of Mission Planner. The examples don't match and a lot of the graphical parameter definitions are missing. The data block display is different and relationship between the drop down box at the top and the parameter selection menu on the right is confusing.
I hope this is the correct place to raise this issue but hope it can be fixed.
Cheers.
Peter

hamishwillee, 2016-12-22:
@Peterrnold  is there a specific page that is affected?

Peterrnold, 2016-12-22:
Hi. 
A lot of it is out of date especially the very basic information. As an example on page
http://ardupilot.org/copter/docs/common-downloading-and-analyzing-data-logs-in-mission-planner.html
Manually review a log the first graphic which explains the log browser is different from the latest version. As an example there is no explanation for the drop down box at the top of the data box and also there is no mention of the parameter menu selection box on the right. I couldn't find the definitions for some of the different parameters.
Another example.
http://ardupilot.org/copter/docs/common-diagnosing-problems-using-logs.html
Eg Compass Interference. The graphic 'Graph This' menu no longer exists? 
 When trying to select tlog’s mag_field (found under “CUSTOM”) . Custom no longer exits.
I may be missing something but after quite a bit of time I couldn't find the information as in the documentation.
I hope this helps.
Peter