ArduPilot/ardupilot_wiki #641: sim_vehicle.py issues

dion-zhang, 2016-12-22:
I am currently following the guide to SITL on http://ardupilot.org/dev/docs/sitl-native-on-windows.html and got to the point after installing FlightGear. However I am stuck when I try to run sim_vehicle on cygwin

leith.hughes.ws@WKS12044 ~/ardupilot/ArduCopter
$ sim_vehicle.py -j4 -L KSFO
SIM_VEHICLE: Start
SIM_VEHICLE: Killing tasks
      1 [main] python2.7 9548 child_info_fork::abort: address space needed by 'cygcrypto-1.0.0.dll' (0x8C0000) is already occupied
SIM_VEHICLE: kill_tasks failed: [Errno 11] Resource temporarily unavailable
SIM_VEHICLE: Starting up at 37.619373,-122.376637,5.3,118 (KSFO)
SIM_VEHICLE: WAF build
SIM_VEHICLE: Configure waf
SIM_VEHICLE: "/home/leith.hughes.ws/ardupilot/modules/waf/waf-light" "configure" "--board" "sitl"
      1 [main] python2.7 10096 child_info_fork::abort: address space needed by 'cygcrypto-1.0.0.dll' (0x8C0000) is already occupied
Traceback (most recent call last):
  File "/home/leith.hughes.ws/ardupilot/Tools/autotest/sim_vehicle.py", line 859, in <module>
    do_build(vehicle_dir, cmd_opts, frame_infos)
  File "/home/leith.hughes.ws/ardupilot/Tools/autotest/sim_vehicle.py", line 465, in do_build
    return do_build_waf(opts, frame_options)
  File "/home/leith.hughes.ws/ardupilot/Tools/autotest/sim_vehicle.py", line 431, in do_build_waf
    run_cmd_blocking("Configure waf", cmd_configure, check=True)
  File "/home/leith.hughes.ws/ardupilot/Tools/autotest/sim_vehicle.py", line 544, in run_cmd_blocking
    p = subprocess.Popen(cmd, **kw)
  File "/usr/lib/python2.7/subprocess.py", line 711, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1235, in _execute_child
    self.pid = os.fork()
OSError: [Errno 11] Resource temporarily unavailable
SIM_VEHICLE: Killing tasks
      1 [main] python2.7 11040 child_info_fork::abort: address space needed by 'cygcrypto-1.0.0.dll' (0x8C0000) is already occupied
SIM_VEHICLE: kill_tasks failed: [Errno 11] Resource temporarily unavailable

I'm not sure why cygcrypto-1.0.0.dll is already occupied.

peterbarker, 2016-12-24:
On Wed, 21 Dec 2016, dion-zhang wrote:

> point after installing FlightGear. However I am stuck when I try to run
> sim_vehicle on cygwin

> 1 [main] python2.7 11040 child_info_fork::abort: address space needed by
> 'cygcrypto-1.0.0.dll' (0x8C0000) is already occupied
> SIM_VEHICLE: kill_tasks failed: [Errno 11] Resource temporarily unavailable

> I'm not sure why cygcrypto-1.0.0.dll is already occupied.

googling tells me this seems to be an issue with the CygWin install. 
Check to make surey ou only have a single CygWin instalation?

auturgy, 2017-11-21:
User support issue rather than a documentation problem, bug report or feature request.

closed 2017-11-21 by auturgy

wntun, 2023-04-12:
I also had the same problem: "SIM_VEHICLE: kill_tasks failed: [Errno 11] Resource temporarily unavailable" issues.
For me, the solution is to turn off the antivirus program.