ArduPilot/ardupilot_wiki #646: RCMAP Input Channel Mapping

deverejohn, 2016-12-29:
[RCMAP Input Channel Mapping](http://ardupilot.org/copter/docs/common-rcmap.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

channel 1 and 2 are swapped

deverejohn, 2016-12-29:
currently reads:
•Channel 1: Roll input
•Channel 2: Pitch input
•Channel 3: Throttle input
•Channel 4: Yaw input

rmackay9, 2016-12-30:
I've changed the RCMAP_ROLL, PITCH, YAW, THROTTLE to be references to the parameters (i.e. so people can click on them and see the parameter description).  I don't think there's an issue with the what is written there for the default though.  input channel 1 is roll by default, input channel 2 is pitch by default.

deverejohn, 2016-12-30:
(I'll also do the pull requests for future items. Thank you!)

Sorry about that, everything starts to blur after a while :-)
I was thinking the parameter order was the correct order, as in the photo.

hamishwillee, 2016-12-30:
That looks good to me. Closing.

closed 2016-12-30 by hamishwillee