ArduPilot/ardupilot_wiki #647: Auxiliary Function Switches ac3.3 only?

deverejohn, 2016-12-29:
[Auxiliary Function Switches](http://ardupilot.org/copter/docs/channel-7-and-8-options.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

there are lots of "AC 3.3 only" notes.  im guessing it should say "AC 3.3 and higher" or similar

rmackay9, 2016-12-30:
Yes, you're right.  If you have a github account, you should be able to click on the "Edit on Github" button on the top right and create a Pull Request to change the wording..

auturgy, 2017-10-15:
The page appears fixed.  I couldn't find a relevant PR to link, unfortunately.  Suggest close.

OXINARF, 2017-10-17:
Yep, @rmackay9 changed this page some time ago. I also don't see any instance of the described problem now, so I'm closing.

closed 2017-10-17 by OXINARF