ArduPilot/ardupilot_wiki #651: Dev: add Contact the developers page

rmackay9, 2017-01-03:
add a "contact the developers" page which lists the various communication methods (gitter, mumble, facebook). We have this info duplicated on various pages within the wiki and the top level ardupilot.org page so a single page listing them would keep the info current without tons of maintenance effort.

auturgy, 2017-11-21:
This is the page http://ardupilot.org/copter/docs/common-contact-us.html
The various related stuff/links in C5 (particularly the dropdown) should be removed I think, and the button on the bottom of http://ardupilot.org/about/team be moved to the top (and maybe landing page). @jpkh @OlivierJB

rmackay9, 2017-11-21:
OK let's close this issue.  other changes can maybe go into a new issue..

closed 2017-11-21 by rmackay9