ArduPilot/ardupilot_wiki #655: Common Autopilots

auturgy, 2017-01-03:
http://ardupilot.org/copter/docs/common-autopilots.html#common-autopilots
needs updating.
I'm happy to do this, but need some background/guidance as to what should/shouldn't be listed.
For instance the CUAV Pixhack is open source, and CUAV is a supporter of Ardupilot, but it isn't listed.
Pixhawk Mini, VRBrain, P9X, Raspilot, BBBMini are also not listed..

hamishwillee, 2017-01-06:
@rmackay9 @tridge Can you provide the necessary guidance?

rmackay9, 2017-01-06:
VRBrain, Raspilot, BBBMini could/should all be added.  We just need someone to take responsibility for creating the pages and maintaining them (not that it takes much maintenance).
I don't think the variants of the Pixhawk should be listed (i.e. Pixhack) because they're primarily just a different board layout from the original. Same goes for the Pixhawk Mini.

rmackay9, 2017-01-06:
For the Pixhawk variants.. one difference is the sensors have been updated.  So perhaps what we could do is add a note at the top of the Pixhawk Overview page to say, "this page covers all the Pixhawk variants including Pixhack, etc".  and then lower down in the Sensors area we provide the list of possible sensors it may be using.  I.e. MPU6k, MPU9250, etc.

Hwurzburg, 2019-10-10:
This has been done ...closing

closed 2019-10-10 by Hwurzburg