ArduPilot/ardupilot_wiki #664: Ardupilot.org: Add pictures to front slider

OlivierJB, 2017-01-06:
Add quality pictures giving a good idea of breadth and depth of Ardupilot.
Make them clickable with link to source and credit, and more information about specific vehicle if available. Point to use cases when applicable.

OlivierJB, 2017-01-13:
Went ahead and uploaded a first batch of new images, and those are now linked to corresponding pages with information on click (except first slide/mosaic). Acquired e-mail permissions for all images found on external commercial websites,  and corresponding pages point to source. For a very few non –commercial (e.g images from individuals on  facebook or from youtube videos) I was not able to get a response from owner and went ahead, will follow-up anyways. 

All images were “verified” to show vehicles specifically running Ardupilot. Pages are still minimal but more info will be added. Another batch of good images is coming hopefully soon, as permissions come in. Still missing Ardusubs in particular. 

Comments, yeahs/nays and suggestions more than welcome of course.

OlivierJB, 2017-01-13:
Removed this image from front page slider: 
[image]

It's a nice image but reverse image search shows same background without copter as stock image, so looks like it is a composite, and no way to verify source, actual copter, FC.

magicrub, 2017-01-13:
do we know where this image came from? maybe someone we know made it

billbonney, 2017-01-14:
It's a composite made by Jani AFAIU. I don't think removing it is a bad idea.

OlivierJB, 2017-01-14:
Oops, could've asked before, I hope Jani won't mind ...  A guess is he had enough work already what with setting up the site in the first place, and not too many pics to choose from. Anyways, can always put it back if there is an outcry,  although I really think composites should be avoided.

Hwurzburg, 2019-10-10:
I think this is completed....closing

closed 2019-10-10 by Hwurzburg