ArduPilot/ardupilot_wiki #665: Ardupilot.org: Populate use cases

OlivierJB, 2017-01-06:
Also integrate info from current use cases section of  Wiki

OlivierJB, 2017-01-23:
I've been making some progress on this  gathering materials. At this time I've thinking of the following (listed here in no particular order:

- **Aerial mapping** (add to existing), possibly splitting it into "3d Mapping" (from wiki use case) and more precise photogrammetry examples, with high precision survey type examples from industry
-  **Search and Rescue** (add to existing)  More on Medical Express 16 and outback challenge, also add [blood sample delivery ](http://www.npr.org/sections/health-shots/2016/09/13/493289511/doctors-test-drones-to-speed-up-delivery-of-lab-tests)in African project; Syria project (unfortunately cancelled)
- **Dispersal of predators** ("bug release", aerobug.au)
- **Mapping for agriculture (NDVI) ** (Agribotix, etc ...)
- **Bathymetry** with rover/boat (Acid lake example, Droneworks LA)
- Education (U. of Kansas, O. in STEM and IEEE, Randy in Japan, others?
- **Wind turbine blade inspection** 
- **Canopy sampling in forestry** (Research, ref:  ICUAS "Canopy Sampling Using an Aerial Manipulator")
- **Lighting management** (Rubidium Rover)
- **Aeromagnetics** (RL, Luis? O., GEM systems)
- **Rover volcano delivery** Randy et al.
- **Cargo Drone and package delivery** (Airbus/Local motor)
- "Fun" **FPV Flying**
- **Photography/Video**. (Since ubiquitous, just a few examples).
- Additional examples  at the research stage.

-

tridge, 2017-01-24:
in the research category, the people using ardupilot fixed wing for flying wind turbines (wind kites) is cute

Hwurzburg, 2019-11-02:
@OlivierJB do you intend to add to this?