ArduPilot/ardupilot_wiki #667: Ardupilot.org

OlivierJB, 2017-01-06:
Write FAQ for visitors completely new to Ardupilot and the world of "drones". Simple and non technical.

What Ardupilot is (and is not), how to get started, etc ... Also sort of and introduction to the wiki that it will point to.

Hwurzburg, 2019-11-02:
@tridge @rmackay9 @OlivierJB  so do we still need this? shouldnt it be part of the Ardupilot.org page...if so its concrete now right? should it be part of the Ardupilot site (Docs target page) as a topic right under HOME in the left side bar?

OlivierJB, 2019-11-04:
Yes (part of ardupilot.org). Still would be one of those  "nice to have" but many edits elsewhere adding details have made this less of an issue.

Hwurzburg, 2026-09-26:
Closing this, as I agree that wiki has improved a lot and FAQs are less needed if the reader follows the "Must read" highlighted sections...

closed 2026-09-26 by Hwurzburg