ArduPilot/ardupilot_wiki #669: SONAR hc sr04

Titisniper71, 2017-01-07:
Hello,
I want to connect a sonar hcsr04 on my apm2.8 in order to be able to drive in indoor.
How do I go about it?
How are connections made?
Is there a specific program?
What other applications can the sonar perform?
  Thank you in advance for the information.
Good evening.
cordially.

hamishwillee, 2017-01-07:
The right place to ask this sort of question is the discuss area: http://discuss.ardupilot.org/

(this area is for docs defects)

closed 2017-01-07 by hamishwillee

rmackay9, 2017-01-09:
It's amazing people are still buying these APM boards.  The last version to run on them is nearly 2 years old (AC3.2.1) and certainly doesn't support object avoidance.  The sonar specified is very simple and can't be used with ardupilot I'm afraid... at least not directly.  With an intermediate companion computer it's possible but it'll never work with an AVR board.