ArduPilot/ardupilot_wiki #676: Advanced Compass Setup

ardisd, 2017-01-12:
[Advanced Compass Setup](http://ardupilot.org/copter/docs/common-compass-setup-advanced.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

On the compass Configuration table it might be useful to add another column for compass 3 and Configuration values for the PIXhawk 2

hamishwillee, 2017-01-30:
@proficnc Are there any special config values for compass (different than original pixhawk) for Pixhawk2?
Does it support an additional compass (ie compass 3?) and is it internal/external?

proficnc, 2017-01-30:
Yes, the Pixhawk 2 has 3 internal compasses, these consist of two MPU9250's (or ICM20948's) and the LSM303D

Externally, the Here has the HMC5983 (it also is fitted with an LIS3MDL but code is deactivated) 

The Here+ RTK has the above compasses, and also a MPU9250 and an AK009916 (a total of 4 compasses) 

There is a new airspeed sensor coming that also has the LIS3MDL onboard.

We will be consolidating down to one Compass per accessory once we have some data from this batch.

auturgy, 2017-10-19:
This page needs a full re-write.

Hwurzburg, 2019-11-02:
there have been a lot of revisions since the last comment...what exactly needs to be added/changed at this point?

Hwurzburg, 2019-12-26:
#2367 pending

Hwurzburg, 2020-01-15:
pr merged....closing

closed 2020-01-15 by Hwurzburg