ArduPilot/ardupilot_wiki #678: Qualcomm Snapdragon Flight Kit feedback

sandeepb614, 2017-01-13:
[Qualcomm Snapdragon Flight Kit](http://ardupilot.org/copter/docs/common-qualcomm-snapdragon-flight-kit.html) 
I work at Qualcomm, looks like you guys are having trouble  with Snapdragon boards, let me know if you need any help with these boards.

hamishwillee, 2017-01-13:
Thanks @sandeepb614 ! @tridge @rmackay9 - please note.

OXINARF, 2017-01-13:
I think the issues with Qualcomm are:

- boards are expensive so only 1 or 2 people in the dev team have them
- input and output isn't as easy as other boards
- the tools license doesn't allow automated testing and binary release

As ArduPilot has a fast development pace it can happen that we break something in these boards as we don't build for them automatically. As very little people have them it usually takes time for someone to notice and then more time for someone with one available to fix it.

rmackay9, 2017-01-25:
@sandeepb614, sorry, I missed this post.

From @OXINARF's list.. I'd really like to emphasize the 3rd point, "the tools license doesn't allow automated testing and binary release".  If there's any thing you could do to allow us to build the binaries on our central build server that would really really help.

It's really hard to get people interested in developing on the board if every single one of them has to build the binary themselves as the very first step.

Can you help with this?

sandeepb614, 2017-01-26:
Is there an private email id I can reach you at?

Sandeep

rmackay9, 2017-01-26:
@sandeepb614, yes, i'm rmackay9 at yahoo.com.

SwiftGust, 2017-03-16:
What I have in trouble is rather HW work.
What troubles me is It requires tons of HW job, when I tried to upgrade Linux in it, it powered out due to  the loose voltage converter(110V -> 220V), and it became brick.
So
1. HW job for factory reset (short tiny small 2 pins) is not easy even with proper soldering tools.
2. Requires antennas for the Wifi/Bluetooth/GPS connection which does not come with the Kit packages.
3. It only have connector for propertiary  ESCs which are also very much expensive.

sandeepb614, 2017-03-16:
Thank you for your feedback.
I will follow up internally.

Sandeep

sandeepb614, 2017-04-21:
**re: concern - Input and output isn't as easy as other boards**
On the hardware side we went through a thorough documentation exercise and posted here on QDN: https://developer.qualcomm.com/hardware/snapdragon-flight/tools 
https://developer.qualcomm.com/download/snapdragon-flight/sdflight-debug-connector-pin-out-hw-user-guide.pdf
Let us know if you need anything specific.

**re: concern - It only have connector for proprietary ESCs which are also very much expensive**
Please refer below documents:
https://developer.qualcomm.com/download/snapdragon-flight/sd-flight-cots-4-in-1-esc-integration.pdf
https://developer.qualcomm.com/download/snapdragon-flight/sdflight-cots-single-esc-integration.pdf

I will get back to you guys on other concerns.

Thanks
Sandeep

OXINARF, 2017-04-22:
@sandeepb614 Thank you for all the documentation and your effort on this. Looking forward for more.

rmackay9, 2017-04-24:
@sandeepb614, on the QC forums, I don't see much information about setting up ardupilot on the snapdragon, it's all just talking about px4.  Is running ardupilot on the snapdragon not something that QC supports?

sandeepb614, 2017-04-25:
Qualcomm snapdragon flight platform is primarily focused on “Snapdragon Navigator” – an advanced flight controller with state of the art technologies. But this is a proprietary software and we don’t have this available to the community yet. As members of Dronecode we do support PX4 development. 
Though Qualcomm development efforts are not focused on Ardupilot, we strongly encourage community to use our flight platform.

re: Adding above links to QC developer forums, we are working on this.

sandeepb614, 2017-04-25:
Please find link to Ardupilot on QC developer page here(under external resources):
https://developer.qualcomm.com/hardware/snapdragon-flight/tools

rmackay9, 2017-04-26:
Thanks very much @sandeepb614.

rmackay9, 2017-04-26:
@sandeepb614,
The next thing would be, do you think it would be OK if we pre-built the ArduPilot binaries for the snapdragon and made them available on our server?  It seems like the answer should be "that's fine" but there was a really crazy long and silly license agreement that everyone needs to sign before downloading the SDK (required to do the build).  That agreement apparently doesn't let us make binaries without getting the OK from QC.

auturgy, 2017-11-21:
Is there an update for this?

khancyr, 2019-01-24:
We drop the support on this board, closing this issue

closed 2019-01-24 by khancyr