ArduPilot/ardupilot_wiki #691: Problem when building ArduPilot for Pixhawk/PX4 on Mac with Make

DonGibson, 2017-01-25:
There is a problem with the problem scripts to build the Copter firmware.  I have not tried compiling for other vehicles.

Following the detailed instructions here (http://ardupilot.org/dev/docs/building-px4-with-make-on-mac.html#building-px4-with-make-on-mac), make fails: 

`Traceback (most recent call last):
  File "/Users/MiniMe/Ardupilot/px4/ardupilot/modules/mavlink//pymavlink/tools/mavgen.py", line 16, in <module>
    from pymavlink.generator import mavgen
  File "/Users/MiniMe/Ardupilot/px4/ardupilot/modules/mavlink/pymavlink/generator/mavgen.py", line 12, in <module>
    from future import standard_library`  
`ImportError: No module named future`  
`mavgen: failed`  
  

The offending line in mavgen.py is

```
from __future__ import print_function
from future import standard_library
standard_library.install_aliases()
from builtins import object
```

There may be a second problem:
`MacbookG3:ArduCopter MiniMe$ make px4-v2-hexa`
`make: *** No rule to make target 'px4-v2-hexa'.  Stop.`

This is a first time installation on a new Mac, and all s/w is up to date including Xcode.

rmackay9, 2017-01-25:
Yes, there was an additional dependency added on the future and lxml python libraries.  I think it can be installed with pip or pip2.

sudo pip2 -q install -U future lxml

rmackay9, 2017-01-25:
By the way, we have moved to waf for all but the windows people.  I plan to update the developer docs in the near-ish future..
https://github.com/ArduPilot/ardupilot/blob/master/BUILD.md

billbonney, 2017-01-25:
@DonGibson I would be waring of user -U flag on OSX, some people are keen to install additions to python in the user space only, but if you have already installed package globally it can be confusing.

my advice is that `sudo pip install lxml future` will work just fine on macOS

DonGibson, 2017-01-25:
Thanks for the help guys. `sudo pip install lxml future` works fine on macOS

OK some progress here. The future and lxml libraries are now present  (in /Library not ~/Library) and the build proceeds until it hits this dependency:

`%%%%
%%%% Building px4io-v2_default in /Users/MiniMe/Ardupilot/px4/ardupilot/modules/PX4Firmware/Build/px4io-v2_default.build/
%%%%
%  PX4_BASE            = /Users/MiniMe/Ardupilot/px4/ardupilot/modules/PX4Firmware/
%  GIT_DESC            = 
%  CONFIG              = px4io-v2_default
%%%%
%%%% Building px4io-v2_default in /Users/MiniMe/Ardupilot/px4/ardupilot/modules/PX4Firmware/Build/px4io-v2_default.build/
%%%%
%  PX4_BASE            = /Users/MiniMe/Ardupilot/px4/ardupilot/modules/PX4Firmware/
%  GIT_DESC            = 
%  CONFIG              = px4io-v2_default
/Users/MiniMe/Ardupilot/px4/ardupilot/modules/PX4Firmware/makefiles//toolchain_gnu-arm-eabi.mk:57: *** Unsupported version of arm-none-eabi-gcc, found: 5.4.1 instead of one in: 4.7.4 4.7.5 4.7.6 4.8.4 4.9.3.  Stop.
make[1]: *** [/Users/MiniMe/Ardupilot/px4/ardupilot/modules/PX4Firmware/Build/px4io-v2_default.build/firmware.px4] Error 2
make: *** [px4-io-v2] Error 2`

billbonney, 2017-01-25:
You need to select the specific version, i'd suggest 4.9 see below 
```
bonney@slimvan:~$ brew search gcc-arm-none-eabi
px4/px4/gcc-arm-none-eabi ✔        px4/px4/gcc-arm-none-eabi-48       px4/px4/gcc-arm-none-eabi-54
px4/px4/gcc-arm-none-eabi-47       px4/px4/gcc-arm-none-eabi-49       px4/px4/gcc-arm-none-eabi-62
bonney@slimvan:~$ brew install px4/px4/gcc-arm-none-eabi-49
==> Installing gcc-arm-none-eabi-49 from px4/px4
==> Downloading https://launchpad.net/gcc-arm-embedded/4.9/4.9-2015-q3-...
```

DonGibson, 2017-01-25:
Wooo !

`%% Copying /Users/dongibson/Ardupilot/px4/ardupilot/modules/PX4Firmware/Images/px4fmu-v2_APM.px4
   text	   data	    bss	    dec	    hex	filename
 931844	   2784	  53864	 988492	  f154c	/Users/MiniMe/Ardupilot/px4/ardupilot/modules/PX4Firmware/Build/px4fmu-v2_APM.build/firmware.elf
PX4 ArduCopter Firmware is in ArduCopter-v2.px4`

So that was successful.  When I tried `make px4-v2-hexa` the error was `make: *** No rule to make target 'px4-v2-hexa'.  Stop.`. ditto with -octa

rmackay9, 2017-01-25:
@DonGibson, ah right.  yes, we've recently combined together many of the frame types.  so instead of building for -hexa you can now set the FRAME_CLASS parameter using a ground station.  Definitely need to update these docs!  I'll get to it eventually..

hamishwillee, 2017-01-30:
@rmackay9 So are all the docs we have currently now up to date for this (ie can this be closed?)

talha72, 2017-08-24:
@rmackay9  so how i can build -Hexa now?

talha72, 2017-08-24:
@rmackay9  i am facing similar problem in windows. how i can install pip2 in windows? i have successfully built '.px4' in "ubuntu" but i cant figure out how to make 'px4-v2-hex' kindly guide.. thanks.

peterbarker, 2017-08-24:
@talha72 Where did you find the instructions specifying `px4-v2-hex`?

That hasn't been a target for sometime.  All frames but heli are now in the one binary, `px4-v2`.

Also, please use the waf build system - `make` has to die!

talha72, 2017-08-24:
@peterbarker thanks a lot sir. it means px4-v2 will work for all? am i right? i am using this "Ardupilot" open source for the first time.  i installed firmware on my Hexa Copter directly from mission planner. it offers separate firmware for quad and hexa copters. so i searched about it and i found "px4-v2-hexa" written in make file example at "Building for Pixhawk/PX4 using Eclipse on Windows".  i read through "px4_targets.mk" file but i couldn't find this target. it made me confused. 

sorry but i couldn't get this "Also, please use the waf build system - make has to die".  i tried to use gitbash with px4console  and px4eclipse. both giving same error.  Can you guide me a little? i shall be thankful to you.  :)