ArduPilot/ardupilot_wiki #693: Navigation Tuning

Winfriedrijssenbeek, 2017-01-27:
[Navigation Tuning](http://ardupilot.org/plane/docs/navigation-tuning.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

Winfriedrijssenbeek, 2017-01-27:
**Accuracy in waypoints!**

Dear guys, we have nicely tuned the new shapes and a beautiful V shape pointed forward. All with the help of Forrest Frantz, who was really excellent in assisting us. Now the next step for our spraying drone is:  how to increase **waypoint accuracy**: we did an experiment today setting the waypoint at 0.5 m and first the drone passed all points and landed safely. Then with some more wind (ques 4m/s) we had it loose the second way point and it stopped there as if in Loiter. We had to set it in Althold and landed safely. We think by studying some of the ardupilot that we would have to cut down on the speed of the craft or increase the way point radius. But the issue is that we plan to fly over crops with good accuracy (0.2 m is acceptable) so when reaching the way point, the drone will approach this with eg speed of 8 m/s, so how can we get it to cross the way point nicely? Any suggestions? Is there a good future in getting better WP accuracy, and yet a drone (octo)  that can fly relatively fast?

PS we noticed today we had 17 Satellites; also interesting we have a RTK piksi system, but that is not too easy to get to work... 

Best Winfried

rmackay9, 2017-01-28:
This is probably a support question rather than a wiki issue.
In any case, if you put a delay on the waypoint command it should pass right through the point.

Winfriedrijssenbeek, 2017-01-30:
We did do some more tests and indeed, with a 2 seconds it looks as if it is more accurate. But what is the experience of people having 17 satellites, what could be the  x y accuracy if these were well used together, and how much would height error be. Best Winfried

rmackay9, 2017-04-25:
This is really a support question so it should probably go in the forums: http://discuss.ardupilot.org/c/arducopter

Sorry to be annoying but if we're not strict, our to-do list turns into a support forum.

closed 2017-04-25 by rmackay9