ArduPilot/ardupilot_wiki #702: bas gps health

lucky1868, 2017-02-03:
[](http://ardupilot.org/plane/search.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

magicrub, 2017-02-03:
Do you have an issue on the wiki to report?

closed 2017-02-03 by magicrub

rmackay9, 2017-02-04:
@lucky1868, you probably meant to raise a support issue?  best to raise these in the support forums.  http://discuss.ardupilot.org/c/arducopter

lucky1868, 2017-02-04:
I have a problem about Bad GPS Health with apm 2.6 arduplane.but it dosent have such problem with arducopter firmware What can I for it?

magicrub, 2017-02-04:
APM 2.6 was depreciated many years ago because there is only do much an 8 it controller can do, no code changes for a long time. Please upgrade to a modern pixhawk.

lucky1868, 2017-02-05:
apm is good enough for me