ArduPilot/ardupilot_wiki #705: Trouble running sim_vehicle.py on Windows 7

sfisher49, 2017-02-06:
Trying to setup the sitl per ardupilot.org.  Was able to resolve the PATH issue per earlier comments, but now everytime I run sim_vehicle.py I get the following message:

$ sim_vehicle.py -j4 -L KSFO
SIM_VEHICLE: Start
SIM_VEHICLE: ** Is (arducopter) really your vehicle type?  Try  -v VEHICLETYPE  if not, or be in the e.g. ArduCopter subdirectory
Traceback (most recent call last):
  File "/home/SFisher/ardupilot/Tools/autotest/sim_vehicle.py", line 829, in <module>
    cmd_opts.frame = default_frame_for_vehicle[cmd_opts.vehicle]
KeyError: 'arducopter'
SIM_VEHICLE: Killing tasks

Ideas???

OXINARF, 2017-02-06:
In what directory are you in? From the message looks to be `arducopter`, which should be actually named `ArduCopter`.

sfisher49, 2017-02-06:
That worked.  thanks.

closed 2017-02-06 by sfisher49