ArduPilot/ardupilot_wiki #707: Setting up XPLANE - SITL on Windows (Failed to load defaults from etc/defaults.parm)

Studentnr1, 2017-02-08:
When trying to setup SITL with Xplane on Windows *cygwin, I always get this message,
despite putting the parameters file into the etc folder, & adding the PATH line to the .bashrc file..
Normal SITL works, but SITL with Xplane won't work...
Do you have any suggestion? 

```
$ ./ArduPlane.elf --home -35,149,584,270 --model xplane
Waiting for XPlane data on UDP port 49001 and sending to port 49000
Started model xplane at -35,149,584,270 at speed 1.0
Starting sketch 'arduplane'
Starting SITL input
bind port 5760 for 0
Serial port 0 on TCP port 5760
Waiting for connection ....
Failed to load defaults from etc/defaults.parm
bind port 5762 for 2
Serial port 2 on TCP port 5762
bind port 5763 for 3
Serial port 3 on TCP port 5763
Closed connection on serial port 0

```

hamishwillee, 2017-02-08:
This area is for questions about the docs. For support queries please see http://discuss.ardupilot.org/

closed 2017-02-08 by hamishwillee

peterbarker, 2017-02-09:
@Studentnr1 The"Failed to load defaults" can be safely ignored.  https://github.com/ArduPilot/ardupilot/pull/5694 should remove that pointless warning.