ArduPilot/ardupilot_wiki #712: Voltage in mission planner

TtrevorT, 2017-02-15:
[](http://ardupilot.org/planner/search.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

OXINARF, 2017-02-15:
Do you have any issue to tell us? Your description is empty and the title isn't descriptive at all.

TtrevorT, 2017-02-15:
Francisco

Thanks for your surprise note.

I started the threat because   I have installed a Pixhawk on a 450 quad and
all is working well , except for the buzzer.  It is silent and I could not
find any information or item in config on how to make the buzzer work.
Then I came across the Lost Plane selection for Ch 7 and all is well.   It
makes the buzzer work , but there is obviously something wrong with the
buzzer, I can hardly hear it.   I put another buzzer on the rx .  It is not
a critical problem,but I thought there must be some way of activating the
buzzer.

I guess my question is, is there a location where you can activate and
change the buzzer for all the other things which it is apparently supposed
to do..... but I am not too worried if it does not?

I am also about to put a Pixhawk in a Bixler just to see how far out I can
safely go.  I am using a RadioLink tx and their little telemetry unit which
sends back the long. and lat. to the tx which could be very useful if it
works properly.

Regards

Trevor
(Canberra )

magicrub, 2017-02-15:
Hi Trevor. This is the wiki issue list which is used for suggesting changes to the documents. if you have a HOWTO question, please post on https://gitter.im/ardupilot/ardupilot or the forums http://discuss.ardupilot.org

closed 2017-02-15 by magicrub

TtrevorT, 2017-02-15:
Thanks Tom.

 Those links will be useful.         I just re-read my email and noticed
that I started a threat (d), which is probably appropriate.  Going down the
park this morning to see how quad flies after a couple of adjustments.

Regards

Trevor

OXINARF, 2017-02-15:
@TtrevorT Like Tom said, please post user issues in the forum. As a correction to what Tom said, please don't ask in Gitter, that's a chat for code development (we have a General room there though where it might be OK to ask).

TtrevorT, 2017-02-16:
Franciso

You guys are the Gods .... we the mere subjects.   I just found out my GPS
was not working because of a loose plug.  All well for the moment.

Kind regards

Trevor

OXINARF, 2017-02-16:
That's a bit much but you can send us a picture of the altar you've made for us :stuck_out_tongue_winking_eye:

Kidding aside, I'm glad you sorted your issue out! Our forum is always open if you have any other doubt.

TtrevorT, 2017-02-16:
Franciso

The Altar...... bit tidier than normal

Trevor