ArduPilot/ardupilot_wiki #721: Mission Planner Language Translations

Bingisser, 2017-02-20:
[Mission Planner Language Translations](http://ardupilot.org/planner/docs/mission-planner-language-translations.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

Added some "Deutsch de-DE" Translations:
[translation.zip](https://github.com/ArduPilot/ardupilot_wiki/files/787287/translation.zip)

narew, 2017-05-31:
Hi, I added the corrected translation of the Polish language:
https://github.com/narew/Mission-planner-polish-language-correction/blob/master/translation.zip

OXINARF, 2017-05-31:
@Bingisser @narew You are in the wrong repository, this one is for documentation updates. For Mission Planner issues you need to open in its repository: https://github.com/ArduPilot/MissionPlanner/issues

@meee1 for your knowledge that this is here.

closed 2017-05-31 by OXINARF

OXINARF, 2017-05-31:
I've added a link to Mission Planner GitHub issues to the Translations page in https://github.com/ArduPilot/ardupilot_wiki/commit/f5c38360676f6e4de28b2a6e63ca10801fdf5ee5

narew, 2017-05-31:
@OXINARF Thanks for arranging. Should I create new issue there like Bingisser did 
https://github.com/ArduPilot/MissionPlanner/issues/1563
or just add my work as the new comment after Him?

OXINARF, 2017-05-31:
@narew Both should work, do as you wish. A new one has the advantage that it brings attention to your contribution, instead of mixing all languages together.