ArduPilot/ardupilot_wiki #724: External LEDs Don't Work on Plane w/ Pixhawk

kirktindel, 2017-02-25:
[External LEDs](http://ardupilot.org/plane/docs/common-external-leds.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

Four different brand new models of Pixhawk, seven different brand new external RGB LED/USB boards, four different i2c splitters and eight new i2c cables.  No matter what I do or how I do it, the external LED shows no signs of life.

Geniune 3DR Pixhawk boards, two HobbyKing HKPilot32 boards, and one unknown manufacturer clone Pixhawk board.  Bixler 3 plane, 3DR power module, 3DR GPS/Compass, 3DR i2c splitter and cables. 3s 4000 mah battery, connected via USB or disconnected from PC, lights only from main Pixhawk board, none from external RGB LED.

Let me know what other info you might need, thanks!

magicrub, 2017-02-25:
What external LED are you referring to?

gmorph, 2017-02-25:
Lets deal with the Pixhawk only as that's what we are most familiar with.  Does the round light on the Pixhawk come on when you apply power?   If not unplug everything from the pixhawk and just plug in a USB cable - if the light doesn't come on AND you have put a stable released firmware on there then your Pixhawk light is bust I'd say.  If it doesn't come on AND you have beta firmware that's probably a software bug and we'd love to know.
Now, when you plug in the external RGB LED plug it straight into the I2C port on the pixhawk i.e. no splitter board and plug in USB power.  When I do this the light on my Pixhawk stops working BUT the external light starts to flash.  If this doesn't work for you and you have stable firmware I'd say your external LED is broken.

Let us know how you go.

Thanks, Grant.

kirktindel, 2017-02-28:
Here's an example of the i2c RGB LED I'm talking about (I have two of these):
https://www.amazon.com/Crius-External-Indicator-Pixhawk-Controller/dp/B00LAV9VLM
Yes, the round light on the Pixhawk does work.  Yellow, blue when armed - it works perfectly.  However the external RGB LED is supposed to take this light's place when plugged in to i2c, and the center round light should cease to work.  This is not the case, the original round LED works, the external does not.

This is an interesting suggestion, gmorph, and to be perfectly honest I had not tried that yet (just plugging it in with no i2c splitter).  I will try this tonight.

Let's assume it works that way, but not with a splitter - what's the solution there?  I have multiple i2c devices I want to use.

Thanks so much, guys!

gmorph, 2017-02-28:
All I2C devices have an address. We expect the external LED to be at a specific address.  Maybe the manufacturer of these devices changed the address?  Can you tell us exactly what vehicle type and Ardupilot software version you have installed on your Pixhawk please?

Thanks, Grant.

rmackay9, 2017-04-25:
Sorry to be annoying but support requests should relaly be raised in the support forums.   http://discuss.ardupilot.org/c/arduplane

This is our wiki to-do list so if there are no objections I will close this issue.

closed 2017-04-25 by rmackay9