ArduPilot/ardupilot_wiki #733: EK2_ALT_SOURCE = 3

deverejohn, 2017-03-09:
[Complete Parameter List](http://ardupilot.org/copter/docs/parameters.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

I tried to find this defined somewhere, (obviously I didn't look at the code) all I found was an old article referring to optical flow (https://groups.google.com/forum/m/#!topic/drones-discuss/LBO8OjCgH74)

"Setting 3 uses the range beacon data."
Is the "range beacon" the same as the rangefinder?
If so, does this setting use the rangefinder as primary alt source?

Thank you!

magicrub, 2017-03-09:
@deverejohn Please see the param descriptions:
http://ardupilot.org/plane/docs/parameters.html#ek2-alt-source-primary-altitude-sensor-source

0	Use Baro
1	Use Range Finder
2	Use GPS
3	Use Range Beacon

option 3 "use range beacon" does not mean rangefinder, that's option 1. If you search "beacon" in the wiki you'll find the [Poszyx page](http://ardupilot.org/plane/docs/common-pozyx.html). However, it would be nice to add that info directly to the EK2_ALT_SOURCE description.

magicrub, 2017-03-09:
Also, I don't see in your link EK2_ALT_SOURCE==3, only EK2_GPS_SOURCE==3 which is different

deverejohn, 2017-03-09:
Tom, thanks for the clarification!
I couldn't figure out what the "beacon" was.
There isn't an option to use the rangefinder (instead of baro) without optical flow, then, from what I understand.
One of the reasons I was looking at this was landing detection.  A while back, on 3.4, I landed at the home location, but it wouldn't stop/disarm. I eventually landed somewhere else.. don't have the log of course.  

I'll post more in the forums instead of here.. thank you.
John

http://ardupilot.org/plane/docs/parameters.html#ek2-alt-source-primary-altitude-sensor-source

closed 2017-03-09 by OXINARF