ArduPilot/ardupilot_wiki #735: Gazebo Installation Outdated

SwiftGust, 2017-03-16:
Hello,
I tried Gazebo Installation on my new laptop with Ubuntu 14.04 LTS 
followed as document here : http://ardupilot.org/dev/docs/using-gazebo-simulator-with-sitl.html#using-gazebo-simulator-with-sitl

Major Number of packages are missing so it's not possible to build from source/ pre-build installation
which does not work is as follows
Build from Source:
Ignition msgs/SD Format/Gazebo and dependency packages

Install from Pre-built :
Number of ROS-Kinetic Related Packages

are missing / does not work

SwiftGust, 2017-03-16:
[image]
It looks like this as I refer here https://github.com/osrf/uctf/tree/master/doc/install_binary

SwiftGust, 2017-03-17:
[image]
in case of ignition msgs, cmake compiling is fine but, 
compiling error comes up when type in make -j4
seems like code missing few things

welkerCode, 2017-10-11:
Has this issue been solved?  I am trying to do the same exact thing on Ubuntu 16.04, and after trying to make (cmake ../ from the tutorial) Gazebo, it gives the following build errors:

-- BUILD WARNINGS
-- 	DART not found, for dart physics engine option, please install libdart-core4-dev.
-- 	libavdevice not found. Recording to a video device will be disabled.
-- 	Player not found, gazebo plugin for player will not be built.
-- 	Oculus Rift support will be disabled.
-- 	ronn not found, manpages won't be generated
-- 	Missing: Ignition Transport (libignition-transport-dev or libignition-transport2-dev)
-- 	OpenSource Virtual Reality (OSVR) support will be disabled.
-- END BUILD WARNINGS

-- BUILD ERRORS: These must be resolved before compiling.
-- 	Missing: SDF version >=4.1.0. Required for reading and writing SDF files.
-- 	Missing: Ignition math2 library.
-- END BUILD ERRORS

CMake Error at CMakeLists.txt:297 (message):
  Errors encountered in build.  Please see the BUILD ERRORS above.

I followed the example of making SDF on the tutorial, but that says that SDF might be outdated and broken.  Have you posted a fix somewhere in the past?

SwiftGust, 2017-10-18:
@welkerCode new tutorial is available https://github.com/SwiftGust/ardupilot_gazebo

closed 2017-10-18 by SwiftGust