ArduPilot/ardupilot_wiki #747: SITL instalation problem

ElefLee, 2017-03-23:
Dear friends
I'm trying to install SITL on Windows 7. The process is getting according to the instructions in:
http://ardupilot.org/dev/docs/sitl-native-on-windows.html#sitl-native-on-windows
until the final step:
**Download and make ArduPilot**
then I get the following results:

USER@USER-PC ~
$ export PATH=$PATH:$HOAM/ardupilot/Tools/autotest

USER@USER-PC ~
$ sim_vehicle.py -j4 --map
-bash: sim_vehicle.py: command not found

USER@USER-PC ~
$ cd/cygdrive/c/USER@USER-PC ~/GitHub/ardupilot
-bash: cd/cygdrive/c/USER@USER-PC: No such file or directory

USER@USER-PC ~
$ cd ~/ardupilot/Arducopter

USER@USER-PC ~/ardupilot/Arducopter
$ make sitl -j4
../mk/apm.mk:24: (x86)/home/USER/ardupilot/mk/sketch_sources.mk: No such file or directory
make: *** No rule to make target '(x86)/home/USER/ardupilot/mk/sketch_sources.mk'.  Stop.

USER@USER-PC ~/ardupilot/Arducopter
$
#  Can somebody please help me on what is going wrong??

magicrub, 2017-03-23:
Hi @ElefLee ! Thank you for trying ArduPilot's SITL.

You're reported a Help question in the developer's bug tracking and feature request system. Have you tried the forums yet? discuss.ardupilot.org

Looking at the wiki, it looks like you're not even following it. Your first line "export PATH.." is supposed to be in the file .bashrc, not executed at the prompt. It's also spelled wrong. In general, if you get an error you should stop and fix it. Plowing forward with more commands will just create more errors.

closed 2017-03-23 by magicrub

ElefLee, 2017-03-23:
Dear magicrub, thank you for your prompt response! I'm new in the community and maybe my query to be ingenuous. You are right about the wrong spelling in the first line. I have also spotted it (somehow later!). As regard the comment: "Your first line "export PATH.." is supposed to be in the file .bashrc, not executed at the prompt." may be I didn't followed the instuctions correctly. thank you again. I will try the  forum discuss.ardupilot.org.