ArduPilot/ardupilot_wiki #781: Tarot Gimbal

chris669, 2017-04-24:
[Tarot Gimbal](http://ardupilot.org/copter/docs/common-tarot-gimbal.html) - Will this procedure be adequate with other Tarot Gimbal? I have the 3 axes : Tarot TL3T01 3-axis or something else I missed in the forum? Sorry if I do not post in the right place. Using a Pixracer, with no Aux out, but main out 5 and 6 free.

rmackay9, 2017-04-24:
@chris669, I haven't tested with a 3-axis tarot.  I expect it will mostly work but the yaw control is the big question.  To allow controlling the yaw, I think you'll want to set-up the "Pan" section on the "Camera Gimbal" page in MP and also probably connect another AUX output from the back of the flight controller to one of the input channels on the gimbal controller.
I really don't know how the gimbal is going to interpret the pwm inputs it receives though.

chris669, 2017-04-24:
Thanks for answering. Yes, that I really want to do is:  use PWM 5 and 6 outputs of the Pixracer to control mode and camera pitch on the gimbal (I've also seen this request on the discuss.px4.io), but I do not know if at this time someone was able to do that and how to proceed, but I'm patient... ;o)

rmackay9, 2017-04-25:
@chris669, so the various software can be confusing, but to be clear, the ardupilot discuss forum is here: http://discuss.ardupilot.org/c/arducopter
I think you've mentioned the px4 software's forum which is quite different software.

chris669, 2017-05-07:
@rmackay9, yes, quite confusing, but using MP now... When I connect a servo to Aux1 it is responding, but the problem is when I connect the gimbal controller, doesn't work. In the Gimbal setup page, now there is a new combo box, do I select Servo, right?

chris669, 2017-05-07:
Ok, after spending my Sunday on it, the only way for me is to work in sbus mode, I mean connecting with an Y wire sbus from receiver to Pixhawk sbus. Firmware 1.5 is not compatible with Tarot 3D III. One question, via sbus will I be able to use SET_ROI functions in MP? How to configure? Thanks.

KirkKirk, 2017-09-17:
Hi Guys,

Recently I got Tarot tl4d (3 axes) and I was trying to make it to work with ROI in mission or "Point camera Here".
In the created circling Mission with ROI in the center of the circle, the camera is pointing to the ROI in the beginning but after the next waypoint it is pointing to random directions but always tilting to the right angle (altitude).
The same situation is with the manual command "Point Camera Here" where I've noticed that the camera yaw is pointing in the wrong direction but to the right altitude. I've followed all instructions and guidance but without success.
The strange thing is that with manual control (RC ch9 and rc10) it's working properly.
Maybe someone who has this type of gimbal will help me to make it work as expected.
I am using PX4 2.4.8 with AC 3.5.2.

Thanks,
Kirk

Bakkertjuh, 2017-10-03:
Hi kirk,

Could u tell/explain me how you connected the gimbal to the pixhawk?

Greetings,

Cor

gmorph, 2017-10-04:
Its on the ArduPilot documentation wiki.
http://ardupilot.org/copter/docs/common-tarot-gimbal.html
Thanks, Grant.

Bakkertjuh, 2017-10-04:
Thought that was only for the 2 axis tarot gimbal

Greetings

KirkKirk, 2017-10-04:
Hi, Bakkertjuh,

I've used CH9 (Tilt) and CH10 (Pan) on the Pixhawk. Don't forget to connect black wire (ground) to FC and to NOT connect the red wire (+5V).
The GoPro, video transmitter, lights, landing gear, the gimbal and BEC are powered from the additional 3S battery. The BEC is connected on the FC servo rail as a FC +5V redundancy.

In the gimbal software:
- Receiver type: Convention
- Default mode: Pan follow
- Tilt: Channel1
- Pan: Channel2
- Control mode: Angle Control

The last test for ROI I set up CH10 as a passthrough and I have a functional ROI but the drone is pointing on the ROI location and the FC is adjusting the tilt. 
It's true, in this case I can manually change the Pan and it looks OK but what is purpose of having 3 axis gimbal if the drone is flying like a "ballerina" :)

Cheers,
Kirk

Bakkertjuh, 2017-10-10:
Try connecting it to the sbus output on you're pixhawk and set BRD_SBUS_OUT to 1. Then in the gimbal software set receiver type to sbus. I used channels 10 to 12 for the tilt/mode/pan. 

Then in mission planner go to "Camera Gimbal" at initial setup under optional hardware. Asign you're tilt and pan to the channels wich u used in the gimbal software. I used servo12 for tilt and servo10 for pan. Don't forget to asign the right input channel. 

Hope this helps.

Greetings Cor