ArduPilot/ardupilot_wiki #782: Setting up SITL on Linux

KirilaMenski, 2017-04-24:
[Setting up SITL on Linux](http://ardupilot.org/dev/docs/setting-up-sitl-on-linux.html) 

Greatings!
I am trying to set up SITL on Linux Mint 17.1. When I write the command 

> sudo pip install future pymavlink MAVProxy

I get the error:

> error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
> 
> ----------------------------------------
> Cleaning up...
> Command /usr/bin/python -c "import setuptools, > tokenize;__file__='/tmp/pip_build_root/pymavlink/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-rr9vp8-record/install-record.txt --single-version-externally-managed --compile failed with error code 1 in /tmp/pip_build_root/pymavlink
Traceback (most recent call last):
  File "/usr/bin/pip", line 9, in <module>
    load_entry_point('pip==1.5.4', 'console_scripts', 'pip')()
  File "/usr/lib/python2.7/dist-packages/pip/__init__.py", line 235, in main
    return command.main(cmd_args)
  File "/usr/lib/python2.7/dist-packages/pip/basecommand.py", line 161, in main
    text = '\n'.join(complete_log)
UnicodeDecodeError: 'ascii' codec can't decode byte 0xd1 in position 55: ordinal not in range(128)

After then trying to performe command: 

> . ~/.bashrc

get an error file not found.
I also receive an error message:

> You need the latest ArduPilot version of JSBSim installed

update it like this not helping:

> cd jsbsim
git pull
./autogen.sh --enable-libraries
make

I do everything according to the instructions but still get the same errors. 
I would really appreciate if you can help me. Regards!

larsks, 2017-07-10:
It looks like you're asking several things here:

## Dependency installation problems

There's a good chance that there is something unique to your environment that is throwing things off, because these instructions do work. I've tested them on Mint 17.3.  Let's try to narrow things down:

1. Become `root`:

        $ sudo su -
        #

2. Install the `future` module by itself:

        # pip install future

3. Now try installing `pymavlink`:

        # pip install pymavlink

Do you see the same failure? If you do, can you include in this issue the output of the `env` command? I am curious if there is anything in your environment that could result in a `UnicodeDecode` error.

FWIW, this is how I successfully installed the dependencies on a Mint 17.3 system:

    apt-get install python-pip python-dev gcc python-lxml
    pip install future
    pip install pymavlink MAVProxy

## Problems source ~/.bashrc

> > . ~/.bashrc

> get an error file not found.
> I also receive an error message:
> 

The [instructions immediately before that](http://ardupilot.org/dev/docs/setting-up-sitl-on-linux.html#add-some-directories-to-your-search-path) have you modify your `~/.bashrc`, so either you skipped those instructions or you modified the wrong file.  You should be editing the `.bashrc` located in your non-`root` home directory.

## Problems with jsbsim

I don't have any suggestions for this one.

OXINARF, 2017-07-12:
@larsks Wow, thank you very much for the detailed answer! You are giving great help with the wiki.

I think this has been answered in some other place, but in any case this isn't the appropriate one for debugging problems. For development questions please post in the forum (http://discuss.ardupilot.org) or ask in Gitter chat (http://gitter.im/ArduPilot/ardupilot).

closed 2017-07-12 by OXINARF