ArduPilot/ardupilot_wiki #785: Mission Planner - AutoPan

Sas-LAB, 2017-04-26:
Good afternoon, 

First of all wanna thank all the Ardupilot community because what you do is amazing. 
I found there is a bug on the mission planner app. 

Starting Mission planner.
Main Screen 
Center Lower options
Activate AutoPan

If it's activated, the map is not moving following the UAV/Ground Vehicle.
(In previous versions it was moving)

Mission Planner Software:
Build:1.3.46
APM Rover 3.1.2

Thank you very much,
Sascha

Sas-LAB, 2017-04-26:
[image]

Sas-LAB, 2017-04-26:
Sorry the Screenshot is only for reference of the AutoPan option.

jabelone, 2017-06-18:
This is for creating issues specifically about the [ArduPilot Wiki](http://ardupilot.org/ardupilot/index.html).  

If you've found a bug in Mission Planner, @meee1 would probably appreciate it if you created an issue here: [MissionPlanner GitHub Issues](https://github.com/ArduPilot/MissionPlanner/issues).

closed 2026-08-28 by Hwurzburg