ArduPilot/ardupilot_wiki #789: Rangefinders (landing page) - HC-SR04

cmeldas, 2017-05-02:
[Rangefinders (landing page)](http://ardupilot.org/copter/docs/common-rangefinder-landingpage.html)

I made RangeFinder based on HC-SR04 sonar and arduino. Is it possible add it here? Meybe it will help someone...
https://github.com/cmeldas/HC-SR04-RangeFinder

magicrub, 2017-06-05:
We would rather the info be hosted on our own wiki. You're more than welcome to create a PR with those changes to add them to the main wiki.

rmackay9, 2017-06-06:
hmm.. some users bring up this sonar every few months and I normally just tell them that it's not a good sonar to use because it requires too much flight controller CPU to be used.  In some ways, I'd like to push those users towards this solution.

auturgy, 2017-10-15:
I agree with both Tom & Randy: would a short PR to the wiki linking to the source suffice?

OXINARF, 2017-10-18:
I'm generally against relying on external pages for our wiki (as stuff tend to disappear or change URLs), but in this case we'll need to link to the GitHub project anyway since it doesn't make sense for us to host the Arduino code. Maybe add it the same way that the VL503 (I probably have the number wrong) is?

Hwurzburg, 2026-08-11:
The Rangefinders landing page now links to a dedicated HC-SR04 page.

Feel free to reopen if you feel this has not been addressed.