ArduPilot/ardupilot_wiki #804: Arducopter documentation needs info about auto-takeoff

AndKe, 2017-05-12:
There is no documentation on autotakeoff, (as part of mission or otherwise) - nowhere users can see info about required throttle stick position, or how to do it at all

rmackay9, 2017-05-13:
I see some documentation on these pages:

- http://ardupilot.org/copter/docs/auto-mode.html#controls
- http://ardupilot.org/copter/docs/mission-command-list.html#takeoff
- http://ardupilot.org/copter/docs/common-mavlink-mission-command-messages-mav_cmd.html#mav-cmd-nav-takeoff

It could probably be improved but there is some info.  I wonder how we could make it better..

AndKe, 2017-05-13:
The auto-mode.html is closest to be helpful.
"If starting the mission while the copter is on the ground the pilot should ensure the throttle is down, then switch to the Auto flight mode, then raise the throttle. The moment that the throttle is raised above zero, the copter will begin the mission."

There should be some words on arming, I've seen students trying to put in auto, then arm (closely to what would be ArduPlane procedure.)  then fooling around with mission restart in MP to make it autostart after uploading/downloding mission and spawning dialogs. 

I think takeoff desires own page, describing RC method, MP method, MavProxy /mavlink commands required.    Also, somthing that even mention that auto is not armable :)

hamishwillee, 2017-05-14:
I'd say a takeoff and landing doc wouldn't hurt. I'd tend to say it should cover both. 

> Also, somthing that even mention that auto is not armable :)

I don't understand this. AFAIK you can now arm in a mission.

rmackay9, 2017-05-15:
@hamishwillee, actually @AndKe is right - it's not possible to arm in AUTO (in copter anyway).  So normally people should arm in another flight mode (like Loiter) and then switch to Auto and then raise the throttle.
OK, we can add some extra wording.

hamishwillee, 2017-05-15:
OK. I'd suggest:

*  a note in http://ardupilot.org/copter/docs/auto-mode.html that arming has to be done first, and how.
* In http://ardupilot.org/copter/docs/arming_the_motors.html I would explain under the very first heading what arming and disarming actually does (ie turns on the motors). Then under that text, I'd move up the note from the bottom of the arming section "You can only arm or disarm in Stabilize, ACRO, AltHold and Loiter mode." and also add explicitly - you cannot arm in AUTO mode!

PR #874 (2017-07-10): update docs re: auto mode and arming
This commit implements suggestions from @hamishwillee in issue #804.
Specifically, it clarifies that a copter must be armed before engaging
auto mode, and it updates the list of armable modes.

This also removes the phrase "...and the propellers will begin to spin
slowly", because the propellers when armed are still spinning fast
enough to cause injury.

I have elected not to include Guided mode in these changes, even
though it is an armable mode.

Closes #804
commit 776265d9 (2017-07-10): update docs re: auto mode and arming

This commit implements suggestions from @hamishwillee in issue #804.
Specifically, it clarifies that a copter must be armed before engaging
auto mode, and it updates the list of armable modes.

This also removes the phrase "...and the propellers will begin to spin
slowly", because the propellers when armed are still spinning fast
enough to cause injury.

I have elected not to include Guided mode in these changes, even
though it is an armable mode.

Closes #804
closed 2017-07-10 by rmackay9