ArduPilot/ardupilot_wiki #828: AC: info on filter frequencys

AndKe, 2017-06-02:
there should be some information on how to use ATC_RAT_*_FILT and ACCEL_Z_FILT telling how users are expected to know when to rise/lower it.

auturgy, 2017-10-15:
@AndKe I agree - we need to improve the tuning guidance, particularly for smaller copters, where this is more likely to be needed..
@lthall can you help with this? @peterbarker is there a page underway re the new/pending FFT that would be a good place to put this?  If the notch filter comes into master, this will need to be covered too.
Particularly relevant for small, overpowered (<300mm) copters with high kv motors, which are pretty common, and often a source of help requests re tuning.

peterbarker, 2017-10-15:
On Sun, 15 Oct 2017, auturgy wrote:

> @AndKe I agree - we need to improve the tuning guidance, particularly for
> smaller copters, where this is more likely to be needed..
> @lthall can you help with this? @peterbarker is there a page underway re the
> new/pending FFT that would be a good place to put this? If the notch filter

I was putting changes for that under common-measuring-vibration.rst - but 
it looks like I didn't actaully make progress on it!

lthall, 2018-01-11:
I have not made commits to the wiki before but I can look at this. For some reason I can't assign it to myself though.

OXINARF, 2018-01-11:
@lthall Assigned!

auturgy, 2020-05-07:
@lthall @andyp1per I suspect that this is "done" and also overtaken by further development: would be good to close, or add an update.

andyp1per, 2020-05-07:
Yes, @lthall 's new tuning guide fixes this

closed 2020-05-07 by auturgy