ArduPilot/ardupilot_wiki #829: AC: ATC_RAT_*  is appear twice in documentation, with different range

AndKe, 2017-06-02:
see http://ardupilot.org/copter/docs/parameters.html#atc-rat-rll-filt-roll-axis-rate-controller-input-frequency-in-hz

within the page, search for: "ATC_RAT_RLL_FILT"
observe that there are four occurrences, two different range.

auturgy, 2017-07-10:
This is the result of https://github.com/ArduPilot/ardupilot/issues/6164 where heli and multi param lists are both generated and published as the Copter list. Suggest close one of these issues and link to the other.

magicrub, 2017-07-10:
As @auturgy said, the problem shows up on the wiki but the fix is on the ardupilot scripts. I'll go ahead and close this in favor of https://github.com/ArduPilot/ardupilot/issues/6164

closed 2017-07-10 by magicrub