ArduPilot/ardupilot_wiki #848: hi. my am 2.8 dose not match by radio controller AT9

mohsen-far, 2017-06-14:
[](http://ardupilot.org/copter/search.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

mohsen-far, 2017-06-14:
hi my APM 2.8 dode not match by radio controller AT9

jabelone, 2017-06-18:
It's unclear what you mean.  Can you please fix up your spelling/grammar and/or add more detail?

If you meant something like "my APM 2.8 does not match/bind/pair with my AT9 remote controller" then this is the wrong place to post.  This isn't a support forum, it's for creating issues specifically about the [ArduPilot Wiki](http://ardupilot.org/ardupilot/index.html).  

If you need technical support for your (very old/outdated) APM 2.8 flight controller please start a forum thread [here](http://discuss.ardupilot.org/).

auturgy, 2017-07-26:
Suggest close

OXINARF, 2017-07-27:
Closed due to no answer, besides looking to be a support question and not a wiki issue. Thanks @auturgy.

closed 2017-07-27 by OXINARF