ArduPilot/ardupilot_wiki #849: Outdated info

SeagullUAV, 2017-06-17:
Hi,

The following page needs an update: 
http://ardupilot.org/copter/docs/common-camera-shutter-triggering-cables-for-purchase.html

**In regards to "Seagull #REC"**
A user modified (_with good intentions off course_) the listing of Seagull #MAP and changed it to Seagull #REC as a new iteration of #MAP, which is not correct. #REC is a dedicated Sony camera controller.

#REC should be changed to Seagull #MAP2 - which is the new version of the previous #MAP:
https://www.seagulluav.com/product/seagull-map2/

#MAP2 now support 300+ cameras from multiple brands !

*****

**In regards to "SkySight MONO"**
SkySight MONO is nowhere to be purchased anymore - Farsight and dauther company; SkySight was my previous company and it closed back in 05/03/2016 - info from the Danish TAX register:
https://datacvr.virk.dk/data/visenhed?enhedstype=virksomhed&id=36056576&soeg=farsight&type=Alle

Seagull #MAP2 is the replacement for SkySight MONO

Also when visiting www.skysight.eu, you will see a note linking to www.seagulluav.com.

*****

**In regards to "LittleSmartThings"**
LittleSmartThings are no longer available due to "Dissolved after bankruptcy" from the status on the Danish CVR register:
https://datacvr.virk.dk/data/visenhed?enhedstype=virksomhed&id=35038000&soeg=35038000

Their website http://littlesmartthings.com/ is still online, and products can be purchased from their shop, **but no orders are being fulfilled** - we have had numerous of our own customers asking us about LittleSmartThings, since they noticed we were both located in Denmark. These customers had ordered from their website, but never received anything, and claimed their money back through a PayPal  complaint.

There might still be some dealers that stock some of their products - but no support whatever is offered in this regard.

I hope this request will be well received, since I personally don't have any interest in letting good people from the community end up loosing their money on no longer existing companies.

*****

Seagull UAV will soon release a new self-contained precision mapping solution offering RTK and PPK, that is build specifically for the Pixhawk 1 and 2.1 and APM platforms.
We would love to also feature this new system on this page, since we have seen a lot of requests for this kind of system - and now finally are going to release it :)

That was all for now - all the best to the team behind ArduPilot.org !!

Best regards
Rone Scholz-Nielsen
CEO / Founder
Seagull UAV

hamishwillee, 2017-06-18:
Hi Rone

Could you please make a PR with the desired changes - that will make it easier for us to accept/merge in a timely manner.

There is a general wiki editing guide here - http://ardupilot.org/ardupilot/docs/common-wiki_editing_guide.html

Regards
Hamish

larsks, 2017-07-08:
@SeagullUAV do you think you will be able to submit these changes as a pull request?

SeagullUAV, 2017-07-09:
Hi Lars,

This has our utmost priority, but unfortunately we are on a tight schedule to complete the testing of the product mentioned in the last section of our initial post above.
Another thing is my lack of knowledge within Github editing, so I'll need to review the documentation before I can begin the submit of these changes.

We hope to find time for this within the near future - any help would also be Greatly appreciated :)

magicrub, 2017-07-09:
@segullUAV at the top right of any wiki page you see "edit on GitHub". Couch that, edit the page and submit at the bottom

auturgy, 2017-10-15:
@SeagullUAV how is this progressing?