ArduPilot/ardupilot_wiki #852: ERROR: the following packages/stacks could not have their rosdep keys resolved to system dependencies

ELSHAIKHI, 2017-06-19:
Hello ,
I am trying to download the Additional Baxter packages for MoveIt, written entirely in C++ (pick and place tutorial) , I stuck with this error could you please update me how to fix it ?

https://github.com/davetcoleman/baxter_cpp

ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
baxter_moveit_scripts: Cannot locate rosdep definition for [baxter_control]
baxter_pick_place: Cannot locate rosdep definition for [baxter_control]

Regards
Thanks in advanced

Georacer, 2017-06-19:
I believe you are asking at the wrong bug tracker. This is the bug tracker for the ArduPilot autopilot firmware, not a Gazebo-generic site.

ELSHAIKHI, 2017-06-19:
Many thanks for alerting me :)

auturgy, 2017-10-15:
Closing as per George’s comment. Not directly an ArduPilot issue.

closed 2017-10-15 by auturgy