ArduPilot/ardupilot_wiki #854: HI MY NAME IS GEORGE PLAVSIC , I AM DEAF MAN. HOW TO DOWNLOAD FOR MAC OR IPAD ABOUT THIS SKY DRONE FPV 2 ON THE ARDUPILOT CONFIGUARTOR ?

GEORGEPLAVSIC180, 2017-06-19:
[](http://ardupilot.org/planner/search.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

OXINARF, 2017-06-19:
Hi George,

This product isn't sold by ArduPilot, I don't have any knowledge about it. I can only link you to what appears to be the manufacturer's website, https://www.skydrone.aero/products/sky-drone-fpv, where it says:

> The Groundstation to receive Video from the Sky Drone FPV 2 unit runs on Windows and Linux with version for MacOS, Android and iOS in the pipeline to be released in Q2 2017.

So it probably isn't available yet. It says it is compatible with QGroundControl, you might want to try it (it works in multiple platforms): http://qgroundcontrol.com

closed 2017-06-19 by OXINARF

magicrub, 2017-06-20:
wow, even George's *name* is in all caps..