ArduPilot/ardupilot_wiki #856: FrSky Telemetry

kojderek, 2017-06-21:
[FrSky Telemetry](http://ardupilot.org/copter/docs/common-frsky-telemetry.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

Why DIY telemetry cable is connected to Telemetry_2 and CraftTheory Cable is connected to Serial4/5. 
Is there any difference?

tomlauzon, 2017-06-21:
As you can see  in the table [here](http://ardupilot.org/copter/docs/common-frsky-telemetry.html#frsky-telemetry-configuration-in-mission-planner) FrSky telemetry is available on serials 1-4. It's a matter of user preference. It makes no difference. Having the pics show different options is good in my opinion.

tomlauzon, 2017-06-21:
Please close the issue if resolved. Thanks

OXINARF, 2017-06-22:
I think @tomlauzon answered the question and I agree that showing different options is the best - also there are pictures of the C&T cable connected to Telem2 too.

closed 2017-06-22 by OXINARF