ArduPilot/ardupilot_wiki #860: Copter Object Avoidance

grefe, 2017-06-28:
[Copter Object Avoidance](http://ardupilot.org/dev/docs/code-overview-object-avoidance.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

There is no description on how to get the radius window with the red curve for the LIDAR in mission planner.  Can this be added, so I can utilize my LIDAR and see it.

Thank you so much.  Mission Planner is a great tool...keep up the great work.

Sincerely,
William K. Grefe
wgrefe1@san.rr.com

rmackay9, 2017-06-28:
Thanks, Yes, we should mention on that wiki page about the radar view.  I'll add that to my list.
By the way, the view comes up automatically so if object avoidance is working and the GCS is connected to the flight controller somehow (i.e. 900mhz radio, wifi, etc) the radar view should just pop-up.

grefe, 2017-06-28:
Hi emackay9,

I have a 900 Mhz rf design radio connected and it gets the distance number in the log and not as a visual representation.  Thank you for adding it to your list.

Sincerely,
William K. Grefe
wgrefe1@san.rr.com

grefe, 2017-06-30:
Hi rmackay9,

Can you please inform me how to get  radar view.  Because it does not automatically pop up.  

Sincerely,
William K. Grefe

auturgy, 2017-07-10:
G'day William,
this is probably a support question, and better answered on discuss.ardupilot.org so others can benefit too, but I'll answer here.  If you enable PRX_TYPE (http://ardupilot.org/copter/docs/parameters.html#prx-type) and have your sensor properly setup (serial settings etc), then MissionPlanner should auto generate the radar when it receives the relevant messages.  If it doesn't, make sure you have the Advanced View selected (not sure if it makes a difference - haven't unchecked that box for a while...).

auturgy, 2017-07-26:
Should only need to update to latest firmware and MP, which both support the feature, then enable appropriate params.  I've put a small PR here: https://github.com/ArduPilot/ardupilot_wiki/pull/896 that hopefully makes it a bit clearer.

auturgy, 2017-07-28:
https://github.com/ArduPilot/ardupilot_wiki/pull/896 should close this issue

closed 2017-07-28 by OXINARF