ArduPilot/ardupilot_wiki #862: Mission Planner Flight Plan

werkineh, 2017-06-30:
Hi, there I am MSc. computer networking student. by now I am doing my thesis on supersonic speed aircraft, speed range from 1mach to 5 mach. I need to set the future waypoints of the craft, unfortunately, I can't get any model that used to model this and can be integrated with any of networking simulator. can I use the simulation data from Ardudrone map to any of network simulator. if I can which one is the best simulator which accepts the mission based data from this simulation like ns2, opnet, qualnet, which one?can i simulate networking part with ArduDrone map simulator when they communicate? can I make the plane to communicate their mission plan each other on fly? please help me I have two mouths to submit the thesis

larsks, 2017-07-10:
You've submitted this issue against the documentation wiki; I think you may actually want to open this against the ardupilot code itself [here](https://github.com/ardupilot/ardupilot/issues).  Since this isn't really a bug report, you may also (or instead) want to consider asking on [the forum](http://discuss.ardupilot.org/).

OXINARF, 2017-07-12:
@larsks Please don't direct people to ArduPilot issues list unless it is a confirmed bug or a feature request. All user support should go into the forum.

I agree that this doesn't belong here, so I'm closing. It has also been answered in MP Gitter chat.

closed 2017-07-12 by OXINARF