ArduPilot/ardupilot_wiki #864: Building ArduPilot for Pixhawk/PX4 on Windows with Make

Imam24, 2017-07-01:
[Building ArduPilot for Pixhawk/PX4 on Windows with Make](http://ardupilot.org/dev/docs/building-px4-with-make.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

Whenever I use the Make command to compile, I keep getting "make' is not recognized as an internal or external command". Any help is appreciated.

Thanks.

OXINARF, 2017-07-03:
Then you are missing some step, make is definitely included in the Pixhawk Toolchain. Try re-installing it.

OXINARF, 2017-09-13:
No answer after two months and make is definitely in toolchain, so I'm closing this. Feel free to comment if you still have an issue.

closed 2017-09-13 by OXINARF