ArduPilot/ardupilot_wiki #879: Maxbotic I2C Sonar Rangefinder

jwgera, 2017-07-12:
[Maxbotic I2C Sonar Rangefinder] I have installed the MB1242 sonar using the i2c port on Pixhawk.   I have tried everything and can not get it to work properly. Sonar values in mp  status sonarrange seems to provide correct values. However when flying in Altitude Hold mode, the copter climbs  and will not hold altitude. 
[image]

rmackay9, 2017-07-13:
Hi there, so the problem is definitely noise coming into the system.  With the analog sensors it was very important to build a special cable but I don't know if that's important for the I2C versions.
This is the older wiki page that describes how to build the cable for the analog sensor: http://ardupilot.org/copter/docs/sonar.html#sonar

By the way, the issues list is for to-do items for the wiki, support should go to the forums: http://discuss.ardupilot.org/

closed 2017-07-13 by rmackay9

jwgera, 2017-07-13:
I did try to shield the wiring and that did not help. I posted here because I believe it is a software problem. I also tried surrounding the sonar with extended foam  to shield it from noise from the sides.