ArduPilot/ardupilot_wiki #898: Using SITL for ArduPilot Testing

HumamHelfawiOptecks, 2017-07-26:
[Using SITL for ArduPilot Testing](http://ardupilot.org/dev/docs/using-sitl-for-ardupilot-testing.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

The link is broken https://raw.githubusercontent.com/ArduPilot/ardupilot/master/Tools/autotest/win_sitl/InstallCygwinAPM.ps1

magicrub, 2017-07-26:
I don't see that link on that page and the links all look OK on the page they should be on http://ardupilot.org/dev/docs/sitl-native-on-windows.html#sitl-native-on-windows

please specify the URL and surrounding text so we can identify it

HumamHelfawiOptecks, 2017-07-26:
http://ardupilot.org/dev/docs/sitl-native-on-windows.html#installation-steps

OXINARF, 2017-08-09:
Fixed by #904

closed 2017-08-09 by OXINARF