ArduPilot/ardupilot_wiki #907: Passthrough FrSky Telemetry Protocol

chris669, 2017-08-08:
[Passthrough FrSky Telemetry Protocol](http://ardupilot.org/plane/docs/common-frsky-passthrough.html) - SERIAL_PROTOCOL set to 10 does not work for Pixracer. Sensors values are not transmitted, unless you set this parameter to 4.

OXINARF, 2017-08-08:
Setting SERIAL_PROTOCOL to 10 means you are using the Passthrough FrSky Telemetry Protocol, which means there is no need for sensor discovery, you need to use a custom telemetry screen that understands the protocol. The only one I'm aware of is the one sold by the same developer of the protocol, FlightDeck from C&T. If it doesn't work, you have to talk to C&T, that's not an ArduPilot product.

closed 2017-08-08 by OXINARF

chris669, 2017-08-08:
surprised by your answer! So you say that for option 10 to works you need to buy a third-party software in order to display it on the Taranis. So putting parameter on 4 gives the opportunity to get these sensors for free. And it works out of the box with PX4 firmware...

OXINARF, 2017-08-09:
You don't need to buy anything. ArduPilot supports multiple options:

- 3 for FrSky D receivers
- 4 for FrSky receivers with SPort using the standard sensor IDs
- 10 for FrSky receivers with SPort using the custom protocol developed by C&T

The protocol is open: it is documented in the Wiki and you can also look at ArduPilot source code to see what it sends on what IDs. This means you (or anyone else) can develop your own telemetry script that understands this custom protocol. What I said is that the only one I know is available is the one sold by C&T.

As far as I know PX4 firmware only has the equivalent to our options 3 and 4 and no equivalent to our option 10, so I don't understand your comparison, specially when we have more options and they all work out of the box...

chris669, 2017-08-09:
so there is a huge problem with the documentation, where it's clearly written that the only option for FrSky is the option 10. so please read the doc. and try it with a Pixracer...

OXINARF, 2017-08-09:
I'm sorry, but it clearly lists all options: http://ardupilot.org/copter/docs/common-frsky-telemetry.html#frsky-telemetry-configuration-in-mission-planner

Actually the FrSky documentation is one of the most complete we have, with lots of information available.

chris669, 2017-08-09:
please read the chapter Passthrough FrSky Telemetry :
Firmware Configuration

Passthrough FrSky Telemetry is enable by setting SERIAL#_PROTOCOL to 10 using your favorite GCS application.

Wrong, this does not work with a Pixracer !!!! you have to set the parameter on 4...

Buenas noches.

tomlauzon, 2017-08-10:
Hi Chris, Passthrough is option 10. Option 4 is repurposed sensor values (not recommended by OpenTX and FrSky). Anybody can write a script that uses the Passthrough protocol. The protocol is not proprietary. When you use passthrough (option 10) you will not see sensors because those values don't use data slots reserved for sensors (as in protocol 4). You'll notice that you get a lot more information and at a higher rate with passthrough, compared to repurposed telemetry (option 4). We are not aware of Apps other than FlightDeck that us this protocol for now. Hope this helps with the confusion.

chris669, 2017-08-10:
hi, thanks for the answer. So when I set parameter to 10 I should see no sensors, right? But I can see the GPS datas, why only GPS data? This is really confusing.

tomlauzon, 2017-08-10:
That's because the Pixhawk can act as a GPS sensor similar to the actual FrSky GPS sensor, and thus it provides the sensor data in the same format. Other telemetry elements from ArduPilot don't have a direct FrSky equivalent either because these sensors don't exist or the data format used to represent that data is too bulky and wastes precious bandwidth on the limited FrSky link (1200bps). We decided to keep GPS as is, in case you lost your drone. In that case you don't need to run a script, this data is available directly within OpenTX because it's a "sensor". You can just display GPS data in another OpenTX screen. Engineering decision...

chris669, 2017-08-10:
Thanks Tom, this is giving sense now. As I built my own DIY FrSky Telemetry wire, I thought the only way to use this wire was the telemetry Passthrough. Maybe documentation could be improved for this, a warning message or something like that. Thanks. Chris

tomlauzon, 2017-08-10:
Thanks for the feedback. I'll see if I can make an improvement.

OXINARF, 2017-08-14:
@tomlauzon I'll suggest you change *Passthrough FrSky Telemetry Protocol* to *FrSky C&T Telemetry Protocol* or something like that, it should be clear it is a custom protocol.

tomlauzon, 2017-08-29:
Sorry, only seeing this now. We didn't want to make this protocol proprietary, but rather open for anyone to use. However, it makes sense. This is a particular protocol that uses passthrough frsky telemetry. The idea is that it's specific to ArduPilot, because it contains ArduPilot specific messages. Also it's been compressed to save precious bits. At first I wanted to call it ArduPilot Specific, but the name didn't stick. I'll think of a new name. Happy to consider names too...

OXINARF, 2017-08-30:
@tomlauzon I understand your point, but the FrSky DIY IDs don't have specific meanings so anyone can develop a protocol using it. C&T did one and that's fine. I, for example, would have done it differently (just design differences, for example, I wouldn't use compression, although I understand why you did). So I think that linking the option name to the author of the protocol is a good idea.