ArduPilot/ardupilot_wiki #911: SITL Architecture Image

hugo-andrade, 2017-08-09:
Hi all,

I made a figure to my paper to illustrate the SITL Architecture in ArduPlane, In this version of figure in Wiki, the connections are deprecated in old versions. 

I think the image that I made is pretty good, if you want to use in Wiki.

SITL Architecture
------
[image]

Thanks,

[SITL Simulator (Software in the Loop)](http://ardupilot.org/dev/docs/sitl-simulator-software-in-the-loop.html)

gmorph, 2017-08-09:
Similar to the picture on this page
http://ardupilot.org/dev/docs/sitl-simulator-software-in-the-loop.html#sitl-simulator-software-in-the-loop

hugo-andrade, 2017-08-09:
Similar, but the ports and connections type in ArduPlane <-> JSBSim, Serial1 and Serial3 are deprecated in this picture. To develop my work using SITL in ArduPlane, I mapped the connections, cause this picture in link were confusing me.

hamishwillee, 2017-08-13:
So just to be clear, you've removed serial connections from the image because they are deprecated/no longer used in the platform?

I think the image looks really "clean" - which is important for readability.

My concern is that is loses information in the labeling - APM planner is just one GSC that could be used here.  Similarly this is ArduPlane/JBSIM specific, but nice to have images that cover all platforms (like the one Grant points to).

hugo-andrade, 2017-08-14:
Hi @hamishwillee, I expressed myself wrong, sorry.

I made this figure to mapping the network structure of ArduPlane SITL comunnication  (UDP and TCP data only). What I saw when I mapped the connections, is some connection shown in the wiki figure don't fit reality, like the UDP connection ArduPlane <-> JSBsim, and the open TCP 5762 that dont exists in wiki figure, only small issues in my opinion.

I agree with you about the specificity when I put APM planner as GSC and JSBSIM as FDM. This figure is the diagram that I used to solve my SITL problem, but if you like and want, I can generalize the image, changing the names and adding the serial connection labels to be used in wiki.

closed 2026-10-10 by Hwurzburg