ArduPilot/ardupilot_wiki #920: First time setup

vinodph123, 2017-08-21:
Hi,

     It is better if you put First time setup like px4 firmware setup they have given the steps for installation of px4.

Regards
Vinod

OXINARF, 2017-08-21:
@vinodph123 Can you explain better what you want? I don't understand what First Time setup you are referring to, is it installing firmware to a board or installing a dev environment?

vinodph123, 2017-08-23:
It is about installing a dev environment in host system like in px4 firmware they have provided jmavsim for checking the operations

OXINARF, 2017-08-23:
We have an extensive developer wiki (http://ardupilot.org/dev/index.html), including guides to build the code in several operating systems. Have you looked at it?

peterbarker, 2017-08-24:
On Wed, 23 Aug 2017, vinodph123 wrote:

> It is about installing a dev environment in host system like in px4 firmware they have provided jmavsim for checking the operations

We supply a Vagrant virtual machine to get people started quickly.

We have extensive simulation support - look for "sim_vehicle.py" and 
"SITL"

OXINARF, 2017-09-01:
I'm closing this as I believe we have enough documentation and options for the first time setup.

@vinodph123 If there is something specific you think we can improve, please comment here and we'll try to look at it.

closed 2017-09-01 by OXINARF