ArduPilot/ardupilot_wiki #922: Copter Mission Command List

pmshopUSMC, 2017-08-22:
[Copter Mission Command List](http://ardupilot.org/copter/docs/mission-command-list.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

Copter Firmware 3.5.2

Loiter Turns does not function as described in ArduPilot copter guide.

Currently listed:
Turn - the number of full rotations to complete around the point.
Dir 1=CW - the direction to turn around the point. -1 = counter clockwise, +1 = clockwise.
Lat, Lon - the latitude and longitude targets. If left as zero it will circle around the current location.
Alt - the target altitude above home in meters. If left as zero it will circle at the current altitude.

Dir has been removed.
So how do you make the craft go CCW?
-1 turn only stops the mission.
IE: LOITER_TURNS: Turns -1 and Radius 100 will make the craft go out 100m, pause and carry on with the next command.

The part that states "Dir 1=CW - the direction to turn around the point. -1 = counter clockwise, +1 = clockwise." is incorrect information as the "Dir" column has been removed.

http://ardupilot.org/copter/docs/mission-command-list.html#loiter-turns

pmshopUSMC, 2017-09-20:
Once again - bump

pmshopUSMC, 2017-09-25:
And again

Hwurzburg, 2020-02-29:
closing....ancient firmware....if issue exists in current firmware, please enter a code issue

closed 2020-02-29 by Hwurzburg