ArduPilot/ardupilot_wiki #930: Building ArduPilot on Windows with waf and Bash

nikhilgupta280695, 2017-09-02:
Step 9. **sudo add-apt-repository ppa:terry.guo/gcc-arm-embedded** 
has been deprecated and a new version is available [here](https://launchpad.net/gcc-arm-embedded/+announcement/13824) 
Kindly update the step as this repository doesn't get added and causes a lot of errors.

OXINARF, 2017-09-06:
New repository doesn't have GCC 4.9. That instruction is useful for whoever still has Ubuntu Trusty in Windows Linux subsystem. Unfortunately that subsystem can't run 32-bit binaries, so the binary we normally use isn't possible to be used.

For Ubuntu Xenial, you can install the gcc-arm-none-eabi present in Ubuntu repos (although not the official ARM release, it should be close enough), but when Microsoft updates Ubuntu version again, you'll lose that ability again because next Ubuntu versions don't have GCC 4.9 anymore. You can also download the package for Willy (http://ppa.launchpad.net/terry.guo/gcc-arm-embedded/ubuntu/pool/main/g/gcc-arm-none-eabi/gcc-arm-none-eabi_4.9.3.2015q3-1wily1_amd64.deb) and try to install it on Xenial.

In any case, I agree the instructions need to be updated, just not to the new PPA.

jeremyolsen, 2018-01-03:
I get errors using either of the previous suggestions in this post.  Are there any other options or is using WSL a dead end?

auturgy, 2018-01-03:
G’day Jeremy,
I apologise in advance: it’ll be at least a fortnight until I can look into this, as I intentionally left my laptop at home to travel without it (both frustrating and liberating).
Apart from uploading via USB, there should be no restrictions using WSL to compile ArduPilot. Can you list down the steps you’ve taken attempting to compile, and the errors you get?
It’s best to put the details on discuss.ardupilot.org and just add a link in here so I can find it. That way others in the community who may not see things in GitHub can assist.

Regards,

James

jeremyolsen, 2018-01-04:
James,

Cheers!  Thanks for the quick response.  I'll comment here and re-post over at discuss.

It's not critical that the WSL compile work for me.  I'm only utilizing WSL until my TX2 arrives and I can have a dedicated Ubuntu environment.  I have successfully compiled APM via Windows for my Navio2 board in the past.  I'm currently working on a Pixhawk project and am trying to setup a SITL/Gazebo environment.

To begin, the [wiki/docs page](http://ardupilot.org/dev/docs/building-ardupilot-onwindows10.html) is listing the 'terry.guo' gcc package/repository as the package to use.  I get a 'key not found' error initially when adding the repository, and ultimately it cannot find the 'trusty' 4.9 package when I try to `apt-get install`.  As stated in the first post above, this looks to be deprecated.

I tried the first alternative, suggested by nikhilgupta280695, using the gcc 5 via 'team-gcc-arm-embedded/ppa' repository.  I can start the compile, but it fails mid-compile and give the following error:

`[435/732] Compiling libraries/AP_NavEKF3/AP_NavEKF3_VehicleStatus.cpp`
`...`
`task in 'objs/AP_Mount' failed
http://ppa.launchpad.net/terry.guo/gcc-arm-embedded/ubuntu/pool/main/g/gcc-arm-none-eabi/gcc-arm-none-eabi_4.9.3.2015q3-1wily1_amd64.deb
{task 140411771971936: cxx AP_Mount_Alexmos.cpp -> AP_Mount_Alexmos.cpp.0.o}`
`...`

After reading the second post above by OXINARF, seems like there is a hard dependency on gcc 4.9 which might explain this error.

I also tried installing the '[...wily1_amd64.deb](http://ppa.launchpad.net/terry.guo/gcc-arm-embedded/ubuntu/pool/main/g/gcc-arm-none-eabi/gcc-arm-none-eabi_4.9.3.2015q3-1wily1_amd64.deb)' after removing the previous gcc5 packages.  This ultimately installed correctly after having to fill a dependency upon the 'libisl13' package.  I found a suitable .deb file [here](https://launchpad.net/ubuntu/xenial/amd64/libisl13/0.14-2).  When executing the rest of the steps, I get an error that occurs during the ./waf configure step as shown below:
`Checking for 'clang++' (C++ compiler)    : Execution failure: ['/opt/gcc-arm-none-eabi-4_9-2015q3/bin/arm-none-eabi-gcc', '--print-sysroot']
Traceback (most recent call last):
  File "<string>", line 27, in run
  File "/usr/lib/python2.7/subprocess.py", line 711, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1343, in _execute_child
    raise child_exception
OSError: [Errno 8] Exec format error`

I'm stumped at this point as it seems the dependency upon gcc4.9 is really the core issue.  I would love to hear there is a simpler solution if once exists.

Thanks,
Jeremy

auturgy, 2018-01-17:
OK, I've had a quick look and I'm also failing for Nuttx and ChibiOS targets.  Linux targets build fine (different tool chain obviously).
Recommend using the px4 tool chain for now http://ardupilot.org/dev/docs/building-px4-with-make.html

yutaoliu, 2018-07-20:
I built ArduCopter on WSL with GCC 4.9.3 and have been trying to upload to a Pixhawk 2 board - until I came across your post. Is there still a limitation? What would be an alternative? Thanks, yutao

auturgy, 2018-07-21:
I just replied in gitter, but I haven’t been able to get —upload working in WSL either, for the stm32 builds (WSL latest).

yutaoliu, 2018-07-23:
Thanks @auturgy for your reply! Uploading from Mission Planner works, but I have not been able to test it since MAVProxy cannot connect to Pixhawk 2 over the COM port. I am asking about it in the MAVProxy gitter.

Hwurzburg, 2020-03-16:
@yutaoliu @auturgy seems this can be closed now....correct?