ArduPilot/ardupilot_wiki #931: FrSky Telemetry

pmshopUSMC, 2017-09-05:
After upgrading to Copter 3.5.1, fewer sensors show up on Taranis X9D + with Open TX 2.2 loaded.
Example: Copter 3.4.6 produced RSSI, RxBt, VSpd, GPS, Alt, and VFAS.
Copter 3.5.1 produces RSSI, RxBt and GPS only.
Also, the mountain with altitude and voltage does not show up on "The Top Bar" of the Taranis.
Same for 3.5.2

If the protocol is changed from 10 to 4, then Craft & Theory Flight Deck does not initialize.

floaledm, 2017-09-05:
Hi Paul,

Thanks for reporting this. This is normal as VFAS, Alt, and VSpd are already transmitted in the passthrough protocol. This info was redundant and removed to save on bandwidth on the FrSky link. These sensors are discoverable using OpenTX scripts such as the latest version of FlightDeck.

If you have FlightDeck, install the latest version, remove the VFAS, Alt, and VSpd sensors from the list of discovered sensors in OpenTX, turn on your autopilot, and run discovery again. These sensors should reappear and you can then use then as before in the OpenTX environment.

pmshopUSMC, 2017-09-06:
the sensors do not reappear.
Tried it several times on copied aircraft profiles.
Only RSSI, RxBt and GPS populate.

floaledm, 2017-09-06:
Are you using FlightDeck 3.2? You can also write your own scripts to "rediscover" any sensor going in passthrough...

pmshopUSMC, 2017-09-06:
Flight deck 3.2.1 did it.
Everything working as it was before.

floaledm, 2017-09-06:
Hi Paul,

Glad to hear! FYI: You posted this issue in the "ardupilot_wiki" project, whereas it looks to me like it should belong in the "ardupilot" project. Just for reference for next time...

TO ANYONE WITH SUPREME ADMIN POWERS: PLEASE CLOSE THIS ISSUE (resolved). THANKS!

closed 2017-09-06 by proficnc